facebookresearch/OccupancyAnticipation #1: ModuleNotFoundError: No module named 'habitat_extensions'

rpartsey, 2020-09-03:
Hi dear @srama2512 
Thanks for open-sourcing your work!

Code snippets from **Maps for occupancy ground-truth generation** and **Maps for evaluating map quality** sections in the `README.md` result in  `ModuleNotFoundError: No module named 'habitat_extensions'` error.

This happens because `OccupancyAnticipation` directory is missing in the `PATH` (check  `sys.path`) during the runtime.

I've overcome that by adding
```
import sys
sys.path.insert(0, os.getcwd())
```
both to `occant_baselines/generate_topdown_maps/generate_occant_gt_maps.py` and `occant_baselines/generate_topdown_maps/generate_environment_layouts.py`.

My `python` version is `Python 3.6.9`.

srama2512, 2020-09-03:
Whoops. I must have missed it in the instructions. The github repo directory must be added to `PYTHONPATH` in `.bashrc`. Does that fix the issue in your case?

rpartsey, 2020-09-04:
Yes, it will fix the issue.

closed 2020-09-04 by srama2512