facebookresearch/OccupancyAnticipation #2: Train models config directory path

rpartsey, 2020-09-03:
Shouldn't the model config path in the configs table in the `README.md` **Training models** section contain `model_configs` directory as shown below?
| Model          | Config directory                                                   |
|-----------------|-----------------------------------------------------------------|
| ANS(rgb)      | `configs/model_configs/ans_rgb`                        |
| ANS(depth)    | `configs/model_configs/ans_depth`                  |
| OccAnt(rgb)   | `configs/model_configs/occant_rgb`                 |
| OccAnt(depth) | `configs/model_configs/occant_depth`           |
| OccAnt(rgbd)  | `configs/model_configs/occant_rgbd`              |
| OccAnt(GT)    | `configs/model_configs/occant_ground_truth` |

srama2512, 2020-09-03:
The pre-models that are available do not obey this directory structure. The above was just an indication pointing to sample config files for each model. In each case, the paths for the following must be changed based on the user's needs:
```
TENSORBOARD_DIR
EVAL_CKPT_PATH_DIR
VIDEO_DIR
CHECKPOINT_FOLDER
```

rpartsey, 2020-09-04:
Got it.

In my humble opinion, it's a bit confusing because we have the configs directory in the project root.
And I initially thought that the path is specified relative to the `configs` directory.

Maybe it would be better to add some explanation and angle brackets around `config` like this

| Model          | Config directory                                                   |
|-----------------|-----------------------------------------------------------------|
| ANS(rgb)      | `<configs>/ans_rgb`                        |
| ANS(depth)    | `<configs>/ans_depth`                  |
| OccAnt(rgb)   | `<configs>/occant_rgb`                 |
| OccAnt(depth) | `<configs>/occant_depth`           |
| OccAnt(rgbd)  | `<configs>/occant_rgbd`              |
| OccAnt(GT)    | `<configs>/occant_ground_truth` |

where `<config>` is an indication pointing to sample config files for each model.

srama2512, 2020-09-04:
All paths in the repo are relative the main directory. So, the paths specified are valid paths for example config files. Is it an issue with the pre-trained models?

rpartsey, 2020-09-04:
No no, I just got confused, because the paths specified in the **Config directory** column don't exist.

For example, there is a `configs` directory in the project root, but the path `configs/ans_rgb` doesn't exist (but the path `configs/model_configs/ans_rgb` does exist).

That's why I suggested either change it to the correct path or to add an explanation that the `config` is an indication pointing to sample config files for each model.

srama2512, 2020-09-04:
Oh, got it. The `model_configs` was missing. Yes, I will fix this in my next update. Thanks!

closed 2020-09-04 by srama2512