facebookresearch/OccupancyAnticipation #3: torch version

rpartsey, 2020-09-03:
When we follow the `habitat-api` installation instructions https://github.com/facebookresearch/habitat-lab/blob/27483d017210cf710a50ba8061948eab58777202/.circleci/config.yml#L110
then the `torch==1.4.0` will be installed.

But the `OccupancyAnticipation` has the `torch==1.2.0` as it's requirement.

When I use `torch==1.2.0` and try to generate maps I get following error
```
ImportError: .virtualenvs/occupancy_anticipation-env/lib/python3.6/site-packages/torch_scatter/scatter_cpu.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN6caffe28TypeMeta21_typeMetaDataInstanceIN3c108BFloat16EEEPKNS_6detail12TypeMetaDataEv
```
When I use `torch==1.4.0`, then the error doesn't occur, but the warning 
```
Default grid_sample and affine_grid behavior has changed to align_corners=False since 1.3.0. Please specify to align_corners=True if the old behavior is desired.
``` 
is displayed.  And I should add `align_corners=False` to all `grid_sample` and `affine_grid behavior` function calls.

I wonder which `torch` version should be used.

srama2512, 2020-09-03:
This is most likely an issue with the library versions for dependencies of torch-scatter. Here are the versions that currently work for me (same as in requirements.txt):
```
torch                  1.2.0
torch-scatter          1.4.0
torchvision            0.4.0
```
The torch_scatter library must be installed after installing the right version of pytorch with the corresponding CUDA (10.0) and CUDNN (v7.4) versions.

I do realize that this is a lower version of pytorch than required in habitat-lab. The latest habitat-lab version requirements conflicted with my code in terms of the `torch_scatter` package. So, I had to downgrade the torch version to run the code. 
My suggestion is that after installing habitat, you could re-install these specific library versions to run the OccAnt methods.

rpartsey, 2020-09-04:
Reinstalled the same library version as specified above

```
torch                  1.2.0
torch-scatter          1.4.0
torchvision            0.4.0
```
 but still getting the same error
```
ImportError: .virtualenvs/occupancy_anticipation-env/lib/python3.6/site-packages/torch_scatter/scatter_cpu.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN6caffe28TypeMeta21_typeMetaDataInstanceIN3c108BFloat16EEEPKNS_6detail12TypeMetaDataEv
```
I have CUDA (v10.2) and cuDNN (v8.0.3) versions.

I'll continue using `torch==1.4.0` as a workaround.

srama2512, 2020-09-04:
This is likely due to the mismatch in CUDA and cuDNN versions. More details can be found here: https://github.com/rusty1s/pytorch_scatter

I will close this for the moment. Feel free to open it again if any issues persist.

closed 2020-09-04 by srama2512