facebookresearch/OccupancyAnticipation #5: Checkpoints size

rpartsey, 2020-09-04:
@srama2512 Could you, please, also give some insights on why the size of the  checkpoints of the same model type vary a lot

For example, let's consider OccAnt(depth):

| README.md     | Model name      | Training data |       Val SPL       | Checkpoint URL | Config URL |  Checkpoint size     |
|-----------------:|:-----------------:|:-------------:|:------------------:|:--------------:|:----------------:|:----------------:|
| Pretrained models  | OccAnt(depth)   |   Gibson 4+   |      0.912        | [ckpt.11.pth](https://dl.fbaipublicfiles.com/OccupancyAnticipation/pretrained_models/occant_depth/ckpt.11.pth) | [ppo_exploration.yaml](https://dl.fbaipublicfiles.com/OccupancyAnticipation/pretrained_models/occant_depth/ppo_exploration.yaml) |    125 MB   |
|  Habitat Challenge | OccAnt(depth)   |   Gibson 2+   |       0.463       | [ckpt.13.pth](https://dl.fbaipublicfiles.com/OccupancyAnticipation/pretrained_models/occant_depth_ch/ckpt.13.pth) | [ppo_exploration.yaml](https://dl.fbaipublicfiles.com/OccupancyAnticipation/pretrained_models/occant_depth_ch/ppo_navigation_evaluate.yaml) |    94.6 MB        |
|  ECCV | OccAnt(depth)   |    Gibson 4+  |      0.784      | [ckpt.6.pth](https://dl.fbaipublicfiles.com/OccupancyAnticipation/pretrained_models_eccv_2020/pretrained_models/occant_depth/ckpt.6.pth) | [ppo_navigation\_evaluate.yaml](https://dl.fbaipublicfiles.com/OccupancyAnticipation/pretrained_models_eccv_2020/pretrained_models/occant_depth/ppo_navigation_evaluate.yaml) |      6.6 MB     |

Especially, I'm curious about the last row with the checkpoint size of 6.6 MB.

srama2512, 2020-09-04:
Haha, yes. The models have changed over time. The ECCV models were pretty lightweight with the `unet_nsf` value being 16 and the pose estimator being smaller. Since then, I have changed the pose estimator capacity and tuned `unet_nsf` further.

The OccAnt(depth) model in the 1st row is large since `unet_nsf` is 64. If `unet_nsf` is 16 instead, the model would be just 30MB. For the habitat challenge model, while `unet_nsf` is 16, the pose estimator capacity is higher. I hope that explains it. I remember trying a larger capacity UNet for the challenge. It didn't give too many benefits and it slowed down the evaluation. So I just stuck to `unet_nsf = 16`.

rpartsey, 2020-09-04:
@srama2512 Thanks for the explanation :)

Closing the issue as resolved.

closed 2020-09-04 by rpartsey