facebookresearch/OccupancyAnticipation #11: Replicating Habitat Challenge results

rpartsey, 2020-09-12:
Hi dear @srama2512 
I followed your _README.md_ instructions and were able to run _OccupancyAnticipation_ evaluation using specified configs without any issues.

The only thing I'd like to ask you is to share the _Dockerfile_ and the _agent.py_ you used for _evalai_ evaluation.

I tried to create my own _Dockerfile_ (and _agent.py_) but all my attempts to set up the _python_ environment and install _OccupancyAnticipation_ dependencies inside the _docker_ failed.

srama2512, 2020-09-12:
Hi @rpartsey 

Unfortunately, I haven't had the chance to create the challenge evaluation script for the latest version of the code. The habitat-api, habitat-sim versions are also not consistent with the challenge version. So I may not be able to provide the script immediately (juggling with a few deadlines).

Coding up the agent should be relatively easy. The code is very similar to the navigation training script that exists in the repo. The biggest issues I faced was installing the right versions of different libraries and getting CUDA to work. So please let me know if you have any specific errors that you are facing and I can help you resolve it. I can provide an older version of my Dockerfile over email if that helps.

rpartsey, 2020-09-12:
Thanks for the quick reply.
The older version of _Dockerfile_ would be of great help to me.
My email is _partsey2412@gmail.com_.

rpartsey, 2021-01-14:
Resolved. Closing the issue.

closed 2021-01-14 by rpartsey