facebookresearch/OccupancyAnticipation #12: Occupancy estimate

rainbow979, 2020-09-29:
There are some bugs. I found that when calculate xyz position of pixels on the ground. The y were very different. But since pixels are on the ground, the y should be same, right? Another question is about the depth images used in your work. The values of pixels in depth images is the distance between real points and camera of agent. What does the distance define? Is it horizontal distance?

srama2512, 2020-10-03:
Could you be a bit more specific? An easily testable example would be helpful. 

The depth images define the distance along Z-axis from the camera-plane to the 3D point, not the distance from the camera center to the 3D point. Please check https://github.com/facebookresearch/habitat-sim for more details on how they are rendered.

srama2512, 2020-10-19:
Closing this due to lack of activity. @rainbow979 - please feel free to open it you have more questions in this regard.

closed 2020-10-19 by srama2512