facebookresearch/OccupancyAnticipation #18: Questions about local policy

vincent341, 2020-12-16:
Q1. It seems that local policy is trained using Imitation Learning/Behavior cloning. Would you mind letting me know where the training data and labels are from for imitation learning?

Q2. I'm not quite clear about the role that local policy plays. Is it possible for the agent to go to long-term goal directed by A* without using local policy?

srama2512, 2020-12-17:
Q1. You can check out the relevant lines [here](https://github.com/facebookresearch/OccupancyAnticipation/blob/4ec7afdb4092a738f73ce7ef7f038735a40ae78b/occant_baselines/rl/ans.py#L691-L704) for how we generate GT actions for imitation. In practice, I did not find a difference b/w RL and IL training. So I just stuck to RL training.

Q2. The A* planner only gives a short-term goal (x, y) to visit. The local policy is responsible for taking actions to reach that (x, y). We have both learned and heuristic versions of the local policy [here](https://github.com/facebookresearch/OccupancyAnticipation/blob/master/occant_baselines/rl/policy.py).

srama2512, 2020-12-20:
Closing this after communication over email.

closed 2020-12-20 by srama2512