facebookresearch/OccupancyAnticipation #20: A question about computing the area of environments

vincent341, 2021-01-19:
Hello. The paper says 

> "we divide the validation environments into small (area less than 36m^2) and large (area greater than 36m^2) to observe the influence of environment size on results".

Would you mind giving me any hint on computing the area/size of the enviroment? How do you compute and obtain it? Which lines of code in the project are responsible for computing the area of the environment?

Your response would be highly appreciated.

srama2512, 2021-01-22:
Hi. These can be computed using the environment layouts generated in [generate_environment_layouts.py](https://github.com/facebookresearch/OccupancyAnticipation/blob/master/occant_baselines/generate_topdown_maps/generate_environment_layouts.py). 

Area = # free-cells * map_scale^2

We computed this on a per-episode basis rather than a per-scene basis since our navigation was limited to a single floor on which the agent is initialized. An alternate way to do this would be to get the [TopDown map from habitat-lab](https://github.com/facebookresearch/habitat-lab/blob/57a0899cfa3c07ccd865f3fc27bd50793de3b458/habitat/tasks/nav/nav.py#L676) for the floor on which the agent is spawned on and perform a similar computation.

closed 2021-01-22 by srama2512

vincent341, 2021-01-22:
Hi @srama2512 ,

Thanks very much for your reply. By Area, does it mean the navigable area or the total area of the environment?

srama2512, 2021-01-22:
The navigable area (only free cells, not occupied).

vincent341, 2021-01-30:
> The navigable area (only free cells, not occupied).

Thanks so much!