facebookresearch/OccupancyAnticipation #21: Question in updating collision maps in ans.py.

JeremyLinky, 2021-01-22:
The t2 is equal to "global_pose[i, 2].item() - math.pi / 2", and then I visualize the collision map, finding that the collision you set is in front of the "global_pose[i, 2].item() - math.pi / 2"  instead of the direction of agent's pose, which is "global_pose[i, 2].item()".  Can you explain that? Much appreciate!!

srama2512, 2021-01-22:
There are two coordinate systems involved here.  

(1) For the purposes of tracking the agent's 2D pose in the world map, we assume the following conventions: agent starts at the map center --- (0, 0) --- with X as forward, Y as rightward, and heading as positive starting from X to Y. 

(2) We also use the standard image indexing for addressing the 2D map and for setting subgoals for planning. Here, the origin is the top-left corner, i.e., the agent starts at (H/2, W/2) where the map size is (H, W), and X is rightward, Y is downward, and heading is positive from X to Y. 

In the [line](https://github.com/facebookresearch/OccupancyAnticipation/blob/4ec7afdb4092a738f73ce7ef7f038735a40ae78b/occant_baselines/rl/ans.py#L560) you are looking at, we basically convert the agent heading from coordinate system (1) to (2). 
```
                x1, y1 = asnumpy(self.states["prev_map_position"][i]).tolist()
                x2, y2 = asnumpy(self.states["curr_map_position"][i]).tolist()
                t2 = global_pose[i, 2].item() - math.pi / 2
```

JeremyLinky, 2021-01-22:
> There are two coordinate systems involved here.
>
> (1) For the purposes of tracking the agent's 2D pose in the world map, we assume the following conventions: agent starts at the map center --- (0, 0) [...]
Thanks for your reply!

closed 2021-01-22 by JeremyLinky