facebookresearch/OccupancyAnticipation #23: AttributeError: 'NoneType' object has no attribute 'dim'

vincent341, 2021-01-30:
I tried to reproduce the **OccAnt(GT)** results by running
`python -u run.py --exp-config configs/model_configs/occant_ground_truth/ppo_exploration.yaml --run-type train`.

But got the following error
```
Process ForkServerProcess-3:
Traceback (most recent call last):
  File "/opt/conda/envs/habitat/lib/python3.6/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/opt/conda/envs/habitat/lib/python3.6/multiprocessing/process.py", line 93, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/project/occant_baselines/supervised/map_update.py", line 272, in map_update_worker
    losses = map_update_fn(ps_args)
  File "/opt/project/occant_baselines/supervised/map_update.py", line 174, in map_update_fn
    mapper_outputs = mapper(mapper_inputs, method_name="predict_deltas")
  File "/opt/conda/envs/habitat/lib/python3.6/site-packages/torch/nn/modules/module.py", line 547, in __call__
    result = self.forward(*input, **kwargs)
  File "/opt/conda/envs/habitat/lib/python3.6/site-packages/torch/nn/parallel/data_parallel.py", line 150, in forward
    return self.module(*inputs[0], **kwargs[0])
  File "/opt/conda/envs/habitat/lib/python3.6/site-packages/torch/nn/modules/module.py", line 547, in __call__
    result = self.forward(*input, **kwargs)
  File "/opt/project/occant_baselines/rl/policy.py", line 603, in forward
    outputs = self.predict_deltas(*args, **kwargs)
  File "/opt/project/occant_baselines/rl/policy.py", line 304, in predict_deltas
    pu_outputs = self.projection_unit(pu_inputs)
  File "/opt/conda/envs/habitat/lib/python3.6/site-packages/torch/nn/modules/module.py", line 547, in __call__
    result = self.forward(*input, **kwargs)
  File "/opt/project/occant_baselines/rl/policy_utils.py", line 57, in forward
    x_full[k], size=(self.V, self.V), mode="bilinear"
  File "/opt/conda/envs/habitat/lib/python3.6/site-packages/torch/nn/functional.py", line 2482, in interpolate
    if input.dim() == 3 and mode == 'nearest':
AttributeError: 'NoneType' object has no attribute 'dim'

I traced the error. It comes from [the following lines](https://github.com/facebookresearch/OccupancyAnticipation/blob/073651bfb66f412a160eddec4574a74b0a3d66a0/occant_baselines/rl/policy_utils.py#L51)
`x_full[k] = F.interpolate(x_full[k], size=(self.V, self.V), mode="bilinear")`

I further print out _x_full[k]_  that caused the error, finding that
`x_full['occ_estimate']=None`

It seems when _k='occ_estimate'_, _x_full['occ_estimate']_, which is _None_, is passed to _F.interpolate()_, causing the error.

Would you please help give any suggestion on why the error occurs and how to solve it?
Many thanks!

```

srama2512, 2021-02-07:
@vincent341  Thank you for pointing this out. I've pushed a fix in the ["fix_occant_ground_truth" baseline](https://github.com/facebookresearch/OccupancyAnticipation/commit/dd9f9e8c886aa346af5b2cb643b73aa212c4697b). I tested this locally and it works. Please let me know if it works for you as well. I'll merge it into master after that.

vincent341, 2021-02-09:
@srama2512 Thanks very much for your quick update. I'm running it. I will let you know the result once it is over.

vincent341, 2021-02-13:
@srama2512 The following training can run smoothly now;
`python -u run.py --exp-config configs/model_configs/occant_ground_truth/ppo_exploration.yaml --run-type train`

But when I tried to evaluate the checkpoint by running the following command;
`python -u run.py --exp-config configs/model_configs/occant_ground_truth/ppo_exploration_eval.yaml --run-type eval`

The **same** error occured again. It was caused by the [same line](https://github.com/facebookresearch/OccupancyAnticipation/blob/073651bfb66f412a160eddec4574a74b0a3d66a0/occant_baselines/rl/policy_utils.py#L51).

**configs/model_configs/occant_ground_truth/ppo_exploration_eval.yaml** was created by myself. It is shown as follows;

Would you mind give any hints on it? Much appreciated.
```
BASE_TASK_CONFIG_PATH: "configs/exploration/gibson_evaluate_noisy.yaml"
TRAINER_NAME: "occant_exp"
ENV_NAME: "ExpRLEnv"
SIMULATOR_GPU_ID: 0
SIMULATOR_GPU_IDS: [0]
TORCH_GPU_ID: 0
VIDEO_OPTION: []
TENSORBOARD_DIR: "tb"
VIDEO_DIR: "video_dir"
EVAL_CKPT_PATH_DIR: "data/new_checkpoints"
NUM_PROCESSES: 1
SENSORS: ["RGB_SENSOR", "DEPTH_SENSOR"]
CHECKPOINT_FOLDER: "data/new_checkpoints"
NUM_EPISODES: 10000
T_EXP: 1000

RL:
  PPO:
    # ppo params
    ppo_epoch: 4
    num_mini_batch: 1
  ANS:
    image_scale_hw: [128, 128]
    MAPPER:
      map_size: 101
      registration_type: 'moving_average'
      label_id: 'ego_map_gt_anticipated'
      ignore_pose_estimator: False
      map_batch_size: 2
      use_data_parallel: True
      replay_size: 10
      gpu_ids: [0]
    OCCUPANCY_ANTICIPATOR:
      type: 'occant_ground_truth'

```

srama2512, 2021-07-01:
Fixed in https://github.com/facebookresearch/OccupancyAnticipation/pull/31

closed 2021-07-01 by srama2512

vincent341, 2021-07-02:
Thanks for the update!

srama2512, 2021-07-02:
Sorry about the latency. Finally got around to fixing it. It was a configuration issue specific to this baseline.