facebookresearch/OccupancyAnticipation #24: Why is the MAX_EPISODE_STEPS set to 1001 instead of 1000?

JeremyLinky, 2021-02-08:
Hi, I noticed that MAX_EPISODE_STEPS is set to 1001 instead of 1000 in the config file during training, but T_EXP is set to 1000, which is not consistent. However, in the test, MAX_EPISODE_STEPS and T_EXP are the same, both are 500. Could you tell me the reason? Thanks!

vincent341, 2021-02-09:
Should we set MAX_EPISODE_STEPS the same value with T_EXP in the test? It seems that once _time step_>_MAX_EPISODE_STEPS_, the algorithm switches to process another episode data in _eval__checkpoint() of occant_exp_trainer.py.

srama2512, 2021-02-10:
Reset is manually called during training. The idea behind 1001 is that the episode should not end within the simulator before it is reset by the training script (see [here](https://github.com/facebookresearch/OccupancyAnticipation/blob/073651bfb66f412a160eddec4574a74b0a3d66a0/occant_baselines/rl/occant_exp_trainer.py#L1288-L1328)). Technically, it can be any value larger than 1000. Nothing special about 1001. This is not done for evaluation though (see [here](https://github.com/facebookresearch/OccupancyAnticipation/blob/073651bfb66f412a160eddec4574a74b0a3d66a0/occant_baselines/rl/occant_exp_trainer.py#L1429-L1431)).

closed 2021-02-10 by srama2512