facebookresearch/OccupancyAnticipation #27: What if train the modules individually?

vincent341, 2021-02-28:
It seems that the method consists of three modules, a mapper/occupancy anticipator, a Global policy and a Local policy. The three modules are trained simultaneously. Would you mind letting me know if it is possible to train the three modules individually? 

Taking training the Global policy for example, replace the outputs of the mapper with the ground truth, the local policy with a heuristic method, and train the global policy. Next freeze parameters of Global policy and train the mapper. At last freeze both the mapper and the Global policy and train the Local Policy. Will it cause any problem if trained in this way? What are advantages of training simultaneously comparing with training individually?

Your response would be highly appreciated.

srama2512, 2021-03-30:
This is currently not supported in the code, but you could try hacking the code easily to try it out. The problem with such a pipeline might be that it involves too many moving parts and it might increase your experimental life-cycle. When compared to the complexities introduced by having such a pipeline, I also don't see a big advantage to doing this when compared to joint training. Right now, it converges reasonably quickly (within 3M frames).

srama2512, 2021-04-22:
Closing this due to lack of activity. Please feel free to open it if you have further questions.

closed 2021-04-22 by srama2512

vincent341, 2021-04-22:
> This is currently not supported in the code, but you could try hacking the code easily to try it out. The problem with such a pipeline might be that it involves too many moving parts and it might [...]

Thanks very much for your explanation!