facebookresearch/OccupancyAnticipation #28: actuation noise

zxwang96, 2021-05-17:
Hi, I was wondering where can I found the noisy task action registry? I only found "MOVE_FORWARD", "TURN_LEFT", "TURN_RIGHT" in the config files and assume they are noisy-free. Thank you.

srama2512, 2021-05-18:
Hi @zxwang96,

You can take a look at [this config file](https://github.com/facebookresearch/OccupancyAnticipation/blob/master/configs/exploration/gibson_evaluate_noisy.yaml) for an example on using noisy actuation. This is implemented in [habitat-lab](https://github.com/facebookresearch/habitat-lab/blob/7964c9ee9fd9bda01ad8709981debb9f408646d6/habitat/sims/habitat_simulator/actions.py#L137-L191).

zxwang96, 2021-05-19:
> Hi @zxwang96,
>
> You can take a look at [this config file](https://github.com/facebookresearch/OccupancyAnticipation/blob/master/configs/exploration/gibson_evaluate_noisy.yaml) for an example on using [...]

Oh, thanks, I see, so you use the built-in implementation of noisy actuation. Then I'd like to ask another question. In your paper, you compared Occant with ANS under noisy situations. But ANS uses collected noise of actuation and sensors as mentioned in their paper. I was wondering how did you justify the comparison result between Occant and ANS with different noise settings? Did you just replace the original custom noise of ANS with the built-in one of habitat-lab? Thank you!

srama2512, 2021-05-19:
Well, the justification is this:  we perform an apples-to-apples comparison under the same conditions (both noisy and noise-free). Since the ANS implementation wasn't available when our work was published, we had to use an approximation. However, I don't expect the results to change much. Our contribution was to improve the mapping while keeping the rest of the components same as the ANS baseline (i.e., our implementation of it).

zxwang96, 2021-05-19:
Gotcha, thanks for your answer! I'll close this issue.

closed 2021-05-19 by zxwang96