facebookresearch/OccupancyAnticipation #29: Mapper training during policy training

fraazor, 2021-05-22:
Hi there,

I have been working on a variation to the projection unit to add a different type of "fog of war" approach for the sensors. However I do not fully understand the code implementation because the mapper training and policy training seem to happen simultaneously. Wouldn't that lead to correlated data/label pairs? Is there some shuffling happening that I am missing?
Would really appreciate an answer to how this was approached.

srama2512, 2021-05-24:
We have a large [replay buffer](https://github.com/facebookresearch/OccupancyAnticipation/blob/073651bfb66f412a160eddec4574a74b0a3d66a0/occant_baselines/common/rollout_storage.py#L265-L368) that stores data over multiple episodes. We randomly sample data from this buffer to break the correlation in data.

fraazor, 2021-05-26:
Thanks a lot. I might have to train them seperately since I have not enough memory for such a large replay buffer.

closed 2021-05-26 by fraazor