facebookresearch/OccupancyAnticipation #30: Not able to load trained models

Singh-sid930, 2021-05-23:
Hi, 
I am trying to load the OccAnt(depth) models but to no luck. Getting the following errors : 
```
Unexpected key(s) in state_dict: "mapper_state_dict", "local_state_dict", "global_state_dict", "config", "extra_state".
```
I am not doing evaluations through your code base but rather trying to replicate the test_occant.py file. 

```
TASK_CONFIG = "OccupancyAnticipation/occant_baselines/config/ppo_exploration.yaml"
        cfg = get_config(TASK_CONFIG)
        occ_cfg = cfg.RL.ANS.OCCUPANCY_ANTICIPATOR
        OccNet = OccAntDepth(occ_cfg)
        OccNet.load_state_dict(torch.load("OccupancyAnticipation/ckpt.11.pth",map_location='cuda:0'))
        OccNet.eval()
````

Is there something I am missing or have to change in the config files in order to do this? I plan not to use the data extracted through Habitat API. I have my own extracted scenes and episodes.

Singh-sid930, 2021-05-23:
Missing whole lot of keys too 
```
Missing key(s) in state_dict: "gp_depth_proj_encoder.inc.conv.conv.0.weight", "gp_depth_proj_encoder.inc.conv.conv.0.bias", "gp_depth_proj_encoder.inc.conv.conv.1.weight", "gp_depth_proj_encoder.inc.conv.conv.1.bias", "gp_depth_proj_encoder.inc.conv.conv.1.running_mean", "gp_depth_proj_encoder.inc.conv.conv.1.running_var", "gp_depth_proj_encoder.inc.conv.conv.3.weight", "gp_depth_proj_encoder.inc.conv.conv.3.bias", "gp_depth_proj_encoder.inc.conv.conv.4.weight", "gp_depth_proj_encoder.inc.conv.conv.4.bias", "gp_depth_proj_encoder.inc.conv.conv.4.running_mean", "gp_depth_proj_encoder.inc.conv.conv.4.running_var", "gp_depth_proj_encoder.down1.mpconv.1.conv.0.weight", "gp_depth_proj_encoder.down1.mpconv.1.conv.0.bias", "gp_depth_proj_encoder.down1.mpconv.1.conv.1.weight", "gp_depth_proj_encoder.down1.mpconv.1.conv.1.bias", "gp_depth_proj_encoder.down1.mpconv.1.conv.1.running_mean", "gp_depth_proj_encoder.down1.mpconv.1.conv.1.running_var", "gp_depth_proj_encoder.down1.mpconv.1.conv.3.weight", "gp_depth_proj_encoder.down1.mpconv.1.conv.3.bias", "gp_depth_proj_encoder.down1.mpconv.1.conv.4.weight", "gp_depth_proj_encoder.down1.mpconv.1.conv.4.bias", "gp_depth_proj_encoder.down1.mpconv.1.conv.4.running_mean", "gp_depth_proj_encoder.down1.mpconv.1.conv.4.running_var", "gp_depth_proj_encoder.down2.mpconv.1.conv.0.weight", "gp_depth_proj_encoder.down2.mpconv.1.conv.0.bias", "gp_depth_proj_encoder.down2.mpconv.1.conv.1.weight", "gp_depth_proj_encoder.down2.mpconv.1.conv.1.bias", "gp_depth_proj_encoder.down2.mpconv.1.conv.1.running_mean", "gp_depth_proj_encoder.down2.mpconv.1.conv.1.running_var", "gp_depth_proj_encoder.down2.mpconv.1.conv.3.weight", "gp_depth_proj_encoder.down2.mpconv.1.conv.3.bias", "gp_depth_proj_encoder.down2.mpconv.1.conv.4.weight", "gp_depth_proj_encoder.down2.mpconv.1.conv.4.bias", "gp_depth_proj_encoder.down2.mpconv.1.conv.4.running_mean", "gp_depth_proj_encoder.down2.mpconv.1.conv.4.running_var", "gp_depth_proj_encoder.down3.mpconv.1.conv.0.weight", "gp_depth_proj_encoder.down3.mpconv.1.conv.0.bias", "gp_depth_proj_encoder.down3.mpconv.1.conv.1.weight", "gp_depth_proj_encoder.down3.mpconv.1.conv.1.bias", "gp_depth_proj_encoder.down3.mpconv.1.conv.1.running_mean", "gp_depth_proj_encoder.down3.mpconv.1.conv.1.running_var", "gp_depth_proj_encoder.down3.mpconv.1.conv.3.weight", "gp_depth_proj_encoder.down3.mpconv.1.conv.3.bias", "gp_depth_proj_encoder.down3.mpconv.1.conv.4.weight", "gp_depth_proj_encoder.down3.mpconv.1.conv.4.bias", "gp_depth_proj_encoder.down3.mpconv.1.conv.4.running_mean", "gp_depth_proj_encoder.down3.mpconv.1.conv.4.running_var", "gp_depth_proj_encoder.down4.mpconv.1.conv.0.weight", "gp_depth_proj_encoder.down4.mpconv.1.conv.0.bias", "gp_depth_proj_encoder.down4.mpconv.1.conv.1.weight", "gp_depth_proj_encoder.down4.mpconv.1.conv.1.bias", "gp_depth_proj_encoder.down4.mpconv.1.conv.1.running_mean", "gp_depth_proj_encoder.down4.mpconv.1.conv.1.running_var", "gp_depth_proj_encoder.down4.mpconv.1.conv.3.weight", "gp_depth_proj_encoder.down4.mpconv.1.conv.3.bias", "gp_depth_proj_encoder.down4.mpconv.1.conv.4.weight", "gp_depth_proj_encoder.down4.mpconv.1.conv.4.bias", "gp_depth_proj_encoder.down4.mpconv.1.conv.4.running_mean", "gp_depth_proj_encoder.down4.mpconv.1.conv.4.running_var", "gp_decoder.up1.conv.conv.0.weight", "gp_decoder.up1.conv.conv.0.bias", "gp_decoder.up1.conv.conv.1.weight", "gp_decoder.up1.conv.conv.1.bias", "gp_decoder.up1.conv.conv.1.running_mean", "gp_decoder.up1.conv.conv.1.running_var", "gp_decoder.up1.conv.conv.3.weight", "gp_decoder.up1.conv.conv.3.bias", "gp_decoder.up1.conv.conv.4.weight", "gp_decoder.up1.conv.conv.4.bias", "gp_decoder.up1.conv.conv.4.running_mean", "gp_decoder.up1.conv.conv.4.running_var", "gp_decoder.up2.conv.conv.0.weight", "gp_decoder.up2.conv.conv.0.bias", "gp_decoder.up2.conv.conv.1.weight", "gp_decoder.up2.conv.conv.1.bias", "gp_decoder.up2.conv.conv.1.running_mean", "gp_decoder.up2.conv.conv.1.running_var", "gp_decoder.up2.conv.conv.3.weight", "gp_decoder.up2.conv.conv.3.bias", "gp_decoder.up2.conv.conv.4.weight", "gp_decoder.up2.conv.conv.4.bias", "gp_decoder.up2.conv.conv.4.running_mean", "gp_decoder.up2.conv.conv.4.running_var", "gp_decoder.up3.conv.conv.0.weight", "gp_decoder.up3.conv.conv.0.bias", "gp_decoder.up3.conv.conv.1.weight", "gp_decoder.up3.conv.conv.1.bias", "gp_decoder.up3.conv.conv.1.running_mean", "gp_decoder.up3.conv.conv.1.running_var", "gp_decoder.up3.conv.conv.3.weight", "gp_decoder.up3.conv.conv.3.bias", "gp_decoder.up3.conv.conv.4.weight", "gp_decoder.up3.conv.conv.4.bias", "gp_decoder.up3.conv.conv.4.running_mean", "gp_decoder.up3.conv.conv.4.running_var", "gp_decoder.up4.conv.conv.0.weight", "gp_decoder.up4.conv.conv.0.bias", "gp_decoder.up4.conv.conv.1.weight", "gp_decoder.up4.conv.conv.1.bias", "gp_decoder.up4.conv.conv.1.running_mean", "gp_decoder.up4.conv.conv.1.running_var", "gp_decoder.up4.conv.conv.3.weight", "gp_decoder.up4.conv.conv.3.bias", "gp_decoder.up4.conv.conv.4.weight", "gp_decoder.up4.conv.conv.4.bias", "gp_decoder.up4.conv.conv.4.running_mean", "gp_decoder.up4.conv.conv.4.running_var", "gp_decoder.outc.conv.weight", "gp_decoder.outc.conv.bias". 
````

srama2512, 2021-05-24:
This is likely because of a mismatch between the model checkpoint and the config. Did you try using this [config](https://github.com/facebookresearch/OccupancyAnticipation/blob/073651bfb66f412a160eddec4574a74b0a3d66a0/configs/model_configs/occant_depth/ppo_exploration.yaml)?

Singh-sid930, 2021-05-26:
So I tried that and it gave me the same error. However I notice something : 
Example of missing keys : 
```
gp_depth_proj_encoder.inc.conv.conv.0.weight", "gp_depth_proj_encoder.inc.conv.conv.0.bias"
```

Example of unexpected keys : 

```
 "mapper.projection_unit.main.main.gp_depth_proj_encoder.inc.conv.conv.0.weight",
```

Seems like `mapper.project_unit.main.main.` is just an additional string attached .

srama2512, 2021-05-30:
@Singh-sid930  - sorry for the late response. The above snippet will not work. You should be following the conventions used in the [trainers](https://github.com/facebookresearch/OccupancyAnticipation/blob/073651bfb66f412a160eddec4574a74b0a3d66a0/occant_baselines/rl/occant_exp_trainer.py) for loading the defining and loading the model. [Here](
https://github.com/facebookresearch/OccupancyAnticipation/blob/073651bfb66f412a160eddec4574a74b0a3d66a0/occant_baselines/rl/occant_exp_trainer.py#L174-L179) is the relevant snippet for why those additional strings are appended to the state_dict. This is the definition of [OccupancyAnticipator](https://github.com/facebookresearch/OccupancyAnticipation/blob/073651bfb66f412a160eddec4574a74b0a3d66a0/occant_baselines/models/occant.py#L381-L403).

Singh-sid930, 2021-05-31:
In the meantime I had got this working by doing this small hack. I could tried the above method which didn't work correctly. High chance I am doing something wrong so I will try it out later.
But just to confirm, the occAntDepth model class pertains to the this model : https://dl.fbaipublicfiles.com/OccupancyAnticipation/pretrained_models/occant_depth_ch/ckpt.13.pth
And inside this model, thee "mapper_state_dict" key is what contains the mapper weights (plus a few more) 
This is my hack : 
```
        occ_cfg = cfg.RL.ANS.OCCUPANCY_ANTICIPATOR
        OccNet = OccAntDepth(occ_cfg).cuda()
        state_dict = torch.load("OccupancyAnticipation/OccDepth.pth", map_location='cuda:0')
        state_dict = state_dict["mapper_state_dict"]

        new_state_dict = OrderedDict()
      
      #### hack
        for k,v in state_dict.items():
            if "mapper.projection_unit.main.main." in k:
                name = k.replace("mapper.projection_unit.main.main.","")
                new_state_dict[name] = v
        OccNet.load_state_dict(new_state_dict)

        OccNet.eval()
```

I just had a few questions about the inpput and output. 

q1: I notice the input is a 128*128 tensor of two channels. 
As I understand it : 
channel 0 = occupancy probability
channel 1 = exploration probability (confidence that it has been explored) 

What is the purpose of exploration? We do not have the exploration probability in our setup. Is it necessary to estimate occupancy? 
What we have is a top down map of occupancy as seen by the robot and we can also accumulate that over steps. 
This map was created from the habitat depth data.

Or can we provide high confidence values to the pixels, that the robot has explore(be it free or occupied)

q2: For the 128,128 map, is the robot ego-centric to the center of the map or the bottom middle of the map ? 

q3 : Is the output also the same semantically as the input in terms of what the channels represent.

srama2512, 2021-06-01:
Yes. That checkpoint corresponds to OccAnt(depth). The method for loading the checkpoint looks good to me. 

q1 - For non-anticipation models, channel 1 indicates whether that part of the space was observed or not (unobserved is 0, observed is 1). For anticipation models, channel 1 indicates the model's confidence about its prediction for that part of the map. Note that an anticipation model does not necessarily have to observe a location to predict its occupancy.

q2 - The robot starts at the center of the map facing upward. But the full map is not rotated as the agent moves.

q3, q6 - What input and output are you referring to here?

Singh-sid930, 2021-06-01:
By input I mean the dict that is to be sent to the model which in the case of OccAntDepth is 
```
input = { 'ego_map_gt' : (bs, 2, H, W) }
```
And output is then 

```
y = model(input)
output = y["occ_estimate"] 
```

q3 : for the output also channel 0 = occupancy probability , channel 1 = prediction confidence ?
q6: in the input of ego_map_gt is the map (the top down occupancy) a snapshot of the current step or the accumulation of all the past steps in the episode. 

q1: quick clarification for the input then, in my ego_gt_map which I send to the OccAntDepth model, channel 1 does not necessarily need to have the any values which represent observation ? Just trying to understand what should I put in channel 1 of the ego_gt_map. We do track explored space, but not confidence of exploration.

srama2512, 2021-06-02:
`ego_map_gt` is basically the geometric projection of the depth sensor readings (non-anticipation). So the q1 response applies.

[Here](https://github.com/facebookresearch/OccupancyAnticipation/blob/073651bfb66f412a160eddec4574a74b0a3d66a0/occant_baselines/rl/occant_exp_trainer.py#L905-L908) is a reference for how `ego_map_gt` is computed.

srama2512, 2021-06-20:
Closing this issue since the questions were addressed. Please feel free to re-open it if there are any more.

closed 2021-06-20 by srama2512