facebookresearch/OccupancyAnticipation #33: What's does height_thresh means when generating map?

AgentEXPL, 2021-08-09:
In habitat_extensions.config.default.py, the following parameters are used. What do they mean?

_C.TASK.GT_EGO_MAP.HEIGHT_THRESH = [0.2, 1.5]
_C.TASK.GT_EGO_WALL_MAP.HEIGHT_THRESH = [0.2, 1.5]
_C.TASK.GT_EGO_MAP_ANTICIPATED.HEIGHT_THRESH = [0.2, 1.5]

srama2512, 2021-08-11:
These thresholds are used to classify what obstacles are in the map. Any point in the point-cloud that has height within these thresholds is an obstacle (for `GT_EGO_MAP`, `GT_EGO_MAP_ANTICIPATED`). In the case of `GT_EGO_WALL_MAP`, the threshold defines what points are walls (not any objects). The default value is not particularly great for the wall. Look [here](https://github.com/facebookresearch/OccupancyAnticipation/blob/aea6a2c0d9701336c01c9c85f5c3b8565d7c52ba/occant_baselines/generate_topdown_maps/configs/occant_gt/gibson_train.yaml#L19-L26) for the values used in the experiments.

AgentEXPL, 2021-08-12:
Thanks a lot for your clear explanation.

closed 2021-08-12 by AgentEXPL