facebookresearch/OccupancyAnticipation #45: mp3d configs

ishikasingh, 2022-01-16:
I was wondering if it is possible to release mp3d exploration eval config files/models as well (those mentioned in the paper)? 
Thanks!

srama2512, 2022-01-25:
Hi @ishikasingh,

You can find the MP3D configs here: https://github.com/facebookresearch/OccupancyAnticipation/tree/main/configs/exploration

We don't train any models on MP3D. We evaluate the Gibson models on MP3D to test cross-dataset transfer.

ishikasingh, 2022-01-26:
I tried using these but the mp3d configs end up running on gibson data. I think you use another file on top of the files in the folder you shared, of the form ppo_exploration.yaml (with some extra params). It would be great if you can share those configs for mp3d!